Bettapro/Solar-Tracer-Blynk-V3 #24: error some identifier and function is undefined/unused

mr12obot, 2021-11-16:
How to fix this error identifier and function is undefined/unused ?

[image]

Bettapro, 2021-11-17:
Could you please share your platformio.ini?

mr12obot, 2021-11-17:
please find below platformio.ini. Is there any missed?

; PlatformIO Project Configuration File

[platformio]
default_envs = esp8266
src_dir = SolarTracerBlynk

[env]
framework = arduino
monitor_speed = 115200
lib_deps = 
    blynkkk/Blynk@^1.0.0
    https://github.com/Bettapro/ModbusMaster.git
    https://github.com/schinken/SimpleTimer.git
    https://github.com/tzapu/WiFiManager.git#2.0.4-beta
    https://github.com/bblanchon/ArduinoJson.git

[extra]
ota_hostname=SOLAR-MODBUS.local
ota_password=admin

;
; [BOARD] ENVIRONMENT DEFS.
;
  
; ESP32 DEV MODULE

# check https://github.com/espressif/arduino-esp32/issues/5436 to use esp232 2.0.0 until 
# it's not ufficially supported by platformio

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
#platform = espressif32
board = esp32dev
platform_packages =
   framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0

[env:esp32dev_ota]
platform = ${env:esp32dev.platform}
extends = env:esp32dev
upload_protocol = espota
upload_port = ${extra.ota_hostname}
upload_flags = --auth=${extra.ota_password}

; WEMOS D1 MINI 32

[env:wemos_d1_mini32]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
#platform = espressif32
board = wemos_d1_mini32
platform_packages =
   framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0

[env:wemos_d1_mini32_ota]
platform = ${env:wemos_d1_mini32.platform}
extends = env:wemos_d1_mini32
upload_protocol = espota
upload_port = ${extra.ota_hostname}
upload_flags = --auth=${extra.ota_password}

; ESP8266

[env:esp8266]
platform = espressif8266
board = esp12e

[env:esp8266_ota]
platform = ${env:esp8266.platform}
extends = env:esp8266
upload_protocol = espota
upload_port = ${extra.ota_hostname}
upload_flags = --auth=${extra.ota_password}

Bettapro, 2021-11-17:
I cannot test it, as I'm not in front of my pc, but I guess doing the following changes will solve.

In _SolarTracerBlynk/SolarTracerBlynk.ino_:

1. Replace line n. 179  `configTime(0, 0, NTP_SERVER_CONNECT_TO);`  with  `configTime(0, 0, NTP_SERVER_CONNECT_TO, CURRENT_TIMEZONE);`
2. Comment lines 180, 181.

Final result should be:
``` 
configTime(0, 0, NTP_SERVER_CONNECT_TO, CURRENT_TIMEZONE);
//setenv("TZ", CURRENT_TIMEZONE, 3);
//tzset();
```

mr12obot, 2021-11-18:
ok it works like a charm. then how about the 2nd log messages below ? it's look like something wrong with library ModbusMaster/src/util

'uint16_t crc16_update(uint16_t, uint8_t)' defined but not used [-Wunused-function]

Bettapro, 2021-11-18:
It's a warning coming from modbusmaster lib., it tells that modbusmaster declared a function inside crc16.h, not used anywhere (that means I can remove it, gaining some extra bytes on flash memory).

Last warning is about a variable not used after its declaration. I guess that part could be coded a little bit better. I already planned to review the procedure on time sync so I will fix it there.

As they are both warnings you can just ignore them and compile the project😉

closed 2021-11-20 by Bettapro