Bettapro/Solar-Tracer-Blynk-V3 #26: Blynk 2.0 support

Bettapro, 2021-11-20:
Should implement blynk 2.0

Bettapro, 2022-01-27:
Thing to do:
- [ ] base dev.
- [ ] template_id/device_name investigation
- [ ] cannot switch from blynk legacy to 2.0 without recompiling

brazzo97, 2022-05-05:
Hi i see in the config file that we can use blynk v2 as an alternative , i was trying to use your program but old blynk closed the sign up and i dont have an account , so i made my local server but it doesnt work great , is it possible to use blynk 2 now thanks

Bettapro, 2022-05-08:
Hello, yes, you can use blynk 1.0 legacy or 2.0 as alternative.
I made few tests and done just the required modification to get it working on 2.0.

Actually I don't really like the way blynk is doing things in 2.0, (too much makers unfriendly), so, as you can see in this issue,it's is still unfinished.
If you are going for 2.0, please keep in mind that you must compile the project on your own, also the whole "blynk web configuration" stuff is up to you. (at least for now and in the nearest future)

Xento, 2022-11-21:
How do I import or create the device in Blynk?

Bettapro, 2022-11-21:
> How do I import or create the device in Blynk?

Refer to Blynk documentation: https://docs.blynk.io/en/
Then, create a discussion if you need further help ( with some details about your problem...)