Bettapro/Solar-Tracer-Blynk-V3 #28: Some RT and ST variables are not ready and not synced!

mr12obot, 2021-11-21:
After compile the project and debug messages via serial is worked until get all values and send updates to blynk.

But please find below the error messages looks like "Some RT and ST variables are not ready and not synced!"

Get all values.
Send updates to Blynk.
Some RT variables are not ready and not synced!
Some ST variables are not ready and not synced!
Starting timed actions...   
SETUP OK!

Then after check via blynk app doesnt appear for this vPIN (min_batt_voltage, max_batt_voltage, charge_status, discharge_status)

How to fix this error issue?

Bettapro, 2021-11-21:
```
Some RT variables are not ready and not synced!
Some ST variables are not ready and not synced!
```
We have some communication issue between esp/solar charge controller, normally, caused by bad wiring or bad solar charge controller not willing so send us the data requested (sometime it happens to me too once in a while, but it solves on its own in few minutes)

Are all other values synced correctly?

First step, I would check the Serial monitor after esp has booted (after "SETUP OK!"), every update request from the controller will generate a message "Update Solar-Tracer SUCCESS!" or "Update Solar-Tracer FAILED!".
Do you have mixed messages? (some fails some success)
Could you wait few secs after boot (10/15sec should be ok) and attach all the serial messages?

(If you're using my blynk project template) you can try with the button "sync all" in settings, it triggers a complete update of all values, so, it could solve.
[image]

mr12obot, 2021-11-21:
After "setup OK!" every update request looks working. The controller generate "Update Solar-Tracer SUCCESS!". 

Then I have no mixed messages or some failed messages. Already wait few secs still doesnt appear for those vPIN mentioned before.

Please find below the capture serial messages.

[image]

Bettapro, 2021-11-21:
In these case: warning messages you got on the startup about the variables not ready and synced are caused by the esp issuing lots of requests to the charge controller (this time in your log al RT variables are synced at boot, only st are not synced).
Should be interesting to check the actual error code of those request...

Did you compile the firmware on just flash the precompiled one?

If you compiled it , check _config.h_ and make sure the related vPins definition are uncommented (otherwise esp wont send any update to blynk about those variables)
```
..
#define vPIN_STAT_ENERGY_GENERATED_TODAY                V15
#define vPIN_STAT_ENERGY_GENERATED_THIS_MONTH           V16
#define vPIN_STAT_ENERGY_GENERATED_THIS_YEAR            V17
..
..
#define vPIN_MIN_BATTERY_VOLTAGE_TODAY                  V19
#define vPIN_MAX_BATTERY_VOLTAGE_TODAY                  V20
..
..
#define vPIN_CHARGING_EQUIPMENT_STATUS_TEXT             V24
#define vPIN_DISCHARGING_EQUIPMENT_STATUS_TEXT          V25
..
```

Bettapro, 2021-11-21:
Actually the modbus return code is managed internally from solar_controller class an it's not exposed.
The only way is to make a temp. change just to have it printed on the serial monitor.

Open ... and add search for `void EPEVERSolarTracer::updateStats()`, and add the Serial.println as below:
```
...
void EPEVERSolarTracer::updateStats()
{
  uint8_t result = this->node.readInputRegisters(MODBUS_ADDRESS_STAT_MAX_PV_VOLTAGE_TODAY, 29);
  
  Serial.println( result );
  
  rs485readSuccess = result == this->node.ku8MBSuccess;
  if (rs485readSuccess)
..
```

this function is called once at the boot, then every hour (more or less) but you can use "sync all" button to trigger it manually.

mr12obot, 2021-11-21:
Since I already try to compare with epever solar station monitor software, the vpin have no issue and appear in their own software normally.

Will try with your suggestion to troubleshoot the return code. For sure will update to you soon after have result.

Bettapro, 2021-11-22:
@mr12obot 
Check latest sources from the repository, now you will get some error codes, eg:
```
Update Solar-Tracer SUCCESS!
Update Solar-Tracer FAILED! [err=226]
Update Solar-Tracer SUCCESS!
Some RT variables are not ready and not synced!
Update Solar-Tracer SUCCESS!
```

Code 226 is a timeout error: no response from the solar controller (busy on other tasks or stuck on some tx/rx buffers..)

I've found one RT variable non synced, that was (probably) the cause for your `Some RT variables are not ready and not synced!
` at boot.
The variable non synced [vPIN_CHARGE_DEVICE_ENABLED] was disabled in my installation, I didn't find any gain to have it enabled (seems it's not doing anything!)

it should solve the issue anyway..

mr12obot, 2021-11-23:
As per using the latest repository and try to use "sync all" button to trigger it manually, now all vPIN appear normally but still have issue that vPIN_MIN_BATTERY_VOLTAGE_TODAY and vPIN_MAX_BATTERY_VOLTAGE_TODAY not appear and still have messages of some RT variables non synced below. Please advice.

```
␄␐�␓␀␃␆#␏␗␗␕␋O�  Get all values
␄␄3␀␀␝?␒␄␃�␃␀␌��␄␄1␀␀␐�o␄␄1␐␀␃\_␄␁␀␀␀␁��␄␄2␀␀␃��  Send updates to Blynk
Some ST variables are not ready and not synced!
Starting timed actions...   
SETUP OK!
----------------------------

␄␄1␀␀␐�oUpdate Solar-Tracer SUCCESS!
␄␁␀␂␀␁\_␄␁␀␀␀␁��␄␄2␀␀␃��Update Solar-Tracer SUCCESS!
␄␄3␛␀␂␎�Update Solar-Tracer SUCCESS!
␄␄1␚␀␂^�Update Solar-Tracer SUCCESS!
␄␄1␐␀␃�gUpdate Solar-Tracer SUCCESS!
␄␄1␀␀␐�oUpdate Solar-Tracer SUCCESS!
␄␁␀␂␀␁\_␄␁␀␀␀␁��␄␄2␀␀␃��Update Solar-Tracer SUCCESS!
␄␄3␀␀␝?␒␄␃�␃␀␌��␄␄1␀␀␐�o␄␄1␐␀␃�␄␄2␀␀␃��␄␄3␛␀␂␎�Update Solar-Tracer SUCCESS!
␄␄1␚␀␂^�Update Solar-Tracer SUCCESS!
␄␄1␐␀␃�gUpdate Solar-Tracer SUCCESS!
␄␄1␀␀␐�oUpdate Solar-Tracer SUCCESS!
␄␁␀␂␀␁\_␄␁␀␀␀␁��␄␄2␀␀␃��Update Solar-Tracer SUCCESS!
```

Bettapro, 2021-11-23:
Could you attach a screenshot of how you set the widget in blynk app? ( these 2 widgets are not included in my template)

battery min/max. and generated energy today/week/month/year/total are requested all at once to the controller, so all of them should be empty.

mr12obot, 2021-11-24:
please find a screenshoot below. i've just reset the generated energy and seems counting normally. but for the min and max battery not appear.

[image]
[image]

[image]
[image]

Bettapro, 2021-11-24:
Checkout my latest code. I think I've disabled them on the controller side at some point, now you should have them updating correctly.

mr12obot, 2021-11-24:
Already add latest code with your update below. but still have messages of some ST variables are not ready not synced and the min and max battery not appear.

Is there any step again to add ?

```
  this->setVariableEnabled(SolarTracerVariables::MAXIMUM_PV_VOLTAGE_TODAY);
  this->setVariableEnabled(SolarTracerVariables::MINIMUM_PV_VOLTAGE_TODAY);
  this->setVariableEnabled(SolarTracerVariables::MAXIMUM_BATTERY_VOLTAGE_TODAY);
  this->setVariableEnabled(SolarTracerVariables::MINIMUM_BATTERY_VOLTAGE_TODAY);
```

Bettapro, 2021-11-24:
Checkout my latest code.
Definetly it will solve, and you'll find some more features you'll like! ( #23 , #29)

mr12obot, 2021-11-25:
Yes, finally the min and max battery voltage normally appear now.

[image]

And the feature debug messages over terminal widget works like charm. but still have messages of some ST variables are not ready not synced. how do we know and tracing ST variables that have an issue ?

[image]

Bettapro, 2021-11-25:
Please share your config.h .

You can also checkout latest code, there're some optimization  that could help and more I've reviewed that warninig to report the numer of un-synced variables.

But, as far as all your data are updated and internal status is 0 (at least after a while) I woundn't be to much concerned.

mr12obot, 2021-11-26:
As per checkout with the latest code, the number ST unsynced variables report now that have 6 ST Var. but still dont know which one.

```
Get all values
Send updates to Blynk
WARNING 6 ST var. are not ready & synced!
Starting timed actions...
SETUP OK!
----------------------------

Update Solar-Tracer SUCCESS!
Update Solar-Tracer SUCCESS!
Update Solar-Tracer SUCCESS!
Update Solar-Tracer SUCCESS!

```

Let you know and please find below the config.h. Please advise.

```
/**
 * Solar Tracer Blynk V3 [https://github.com/Bettapro/Solar-Tracer-Blynk-V3]
 * Copyright (c) 2021 Alberto Bettin 
 *
 * Based on the work of @jaminNZx and @tekk.
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */

/**
 *  
 * Use this file to set parameters and settings for your 
 * setup.
 *
 */

#pragma once

/**
 * DEBUG MESSAGES
 * 
 * Support for debug messages via serial
 */
// specify which Serial to use for debug messages
#define BOARD_DEBUG_SERIAL_STREAM Serial
// set accordingly: monitor_speed in platformio.ini
#define BOARD_DEBUG_SERIAL_STREAM_BAUDRATE 115200

/**
 * STATUS
 * 
 * Define how/if the board should notify an error
 */
// blink a led when errors occur
//#define USE_STATUS_LED
#ifdef USE_STATUS_LED
  //specify pin number connected to the status led
  #define STATUS_LED_PIN 2
[... 163 lines cut ...]

#define vPIN_CONTROLLER_HEATSINK_TEMP                   V62
#define vPIN_CHARGE_DEVICE_ENABLED                      V63 //
#define vPIN_LOAD_CURRENT                               V64
#define vPIN_LOAD_POWER                                 V65
#define vPIN_LOAD_ENABLED                               V66 
#define vPIN_INTERNAL_STATUS                            V67 // internal

#define vPIN_BATTERY_BOOST_VOLTAGE                      V68
#define vPIN_BATTERY_EQUALIZATION_VOLTAGE               V69
#define vPIN_BATTERY_FLOAT_VOLTAGE                      V70
#define vPIN_BATTERY_FLOAT_MIN_VOLTAGE                  V71
#define vPIN_BATTERY_CHARGING_LIMIT_VOLTAGE             V72
#define vPIN_BATTERY_DISCHARGING_LIMIT_VOLTAGE          V73
#define vPIN_BATTERY_LOW_VOLTAGE_DISCONNECT             V74
#define vPIN_BATTERY_LOW_VOLTAGE_RECONNECT              V75
#define vPIN_BATTERY_OVER_VOLTAGE_DISCONNECT            V76
#define vPIN_BATTERY_OVER_VOLTAGE_RECONNECT             V77
#define vPIN_BATTERY_UNDER_VOLTAGE_RESET                V78
#define vPIN_BATTERY_UNDER_VOLTAGE_SET                  V79

#define vPIN_INTERNAL_DEBUG_TERMINAL                    V80 // terminal widget
#define vPIN_UPDATE_ALL_CONTROLLER_DATA                 V81 // update scc data

```

Bettapro, 2021-11-29:
The 6 vars are about energy generated and min/mav voltage during the day (exactly 6 as reported in the message).

I can see that those are working ok after the 2° refresh cycle, then it should be boot related, probably a timeout error generated by the solar charge controller due too much update requests.

Problems like this cannot be solved (if there's a solution actually) because:

1. epever is not providing any clue about this (manuals are not very documented on this topics)
2. communication errors could happen
3. it requires some kind of sniffing of the modbus packets to find something interesting (I don't have time to spend here at the moment)
4. this is not a bug in the code (communication is working fine), just something not behaving as expected, I already encontered some strange implementations here, I won't be surprised if this will added to the list 😆

As the esp should boot only once in a while this issue shouldn't cause too much troubles, I will create an issue for this specific issue and close this.
Keep thing monitored and report if you find any issue.

closed 2021-11-29 by Bettapro

mr12obot, 2021-11-30:
Great explaination. Let us report if find any issue and monitored for the sync error on sync all while booting.