Bettapro/Solar-Tracer-Blynk-V3 #31: solar PV status, device SCC status and battery control parameter

mr12obot, 2021-11-21:
I already try originally software from the Epever solar station monitor via USB cable to PC

Is it possible to have feature for solar PV status, device SCC status and battery control parameter capture below? 

So we can read and write all battery parameter over blynk app.

[image]

[image]

Bettapro, 2021-11-21:
[image]

As far as I can see  solar status, charging status, device status are related to the same variable address:
- charging status tells if controller is charging or not
- solar status tells the status (cut out, float, boost, equalization)
- device status tells if a fault occours

here we have them merged in one:
[image]

- solar controller is not charging ( text: standby)
- solar controller is charging (text: boost/float/equalization)
- fault / not normal state (text: every other text starting with !, eg: "! PV INPUT SHORT")

Battery settings are already in progress, data read from epever controller is done, still need to implement to set them from blynk app (big question on the validation done by controller itself...)
For the moment I will be happy to have all voltage levels + battery capacity. 🤔 
As you can see, I still have to label them all 😆  

[image]

Bettapro, 2021-11-24:
[image]

Implemented most of the battery charge and discharge parameters (read/write).
The other still missing (you first comment) will follow at some point.😉
Controller does not validate them in any way, so proceed with attention.
Sometimes the change requests fails due timeouts, increasing the ms between refreshes helps.
The sync all button comes handy to verify all parameters are set correctly.

mr12obot, 2021-11-25:
Yes, already try this read/write battery parameter is working now and validate with sync all button manually.

Still missing for the Battery Type, Battery Capacity, Boost Duration, Eq Duration and etc.

[image]

mr12obot, 2021-12-06:
Is there any implemented code for the other still missing battery control parameter?

Bettapro, 2021-12-06:
> Is there any implemented code for the other still missing battery control parameter?

Not yet, currently I'm taking a small break from the active dev. of this project. Then v3.0.4 will be my main focus (check milestone 3.0.4).
Then, I think we can say I could plan it for the next version.

You'll be notified as I'll reference this issue.

commit 9bcad3ab (2022-03-03): Add more battery control settings, close #31
closed 2022-03-03 by Bettapro