Bettapro/Solar-Tracer-Blynk-V3 #54: Swapping the minimum and maximum PV_VOLTAGE_TODAY

theaudio, 2022-11-05:
**Describe the bug**
I noticed that these values are thrown in VariableDefiner.cpp

MAXIMUM_PV_VOLTAGE_TODAY
MINIMUM_PV_VOLTAGE_TODAY

**Patch**
-    this->initializeVariable(Variable::MAXIMUM_PV_VOLTAGE_TODAY, "Today PV max. volt.", VariableDatatype::DT_FLOAT, diff --git a/SolarTracerBlynk/src/core/VariableDefiner.cpp b/SolarTracerBlynk/src/core/VariableDefiner.cpp
index 4e984ae..d3bfd30 100644
--- a/SolarTracerBlynk/src/core/VariableDefiner.cpp
+++ b/SolarTracerBlynk/src/core/VariableDefiner.cpp
@@ -46,8 +46,8 @@ VariableDefiner::VariableDefiner()
     this->initializeVariable(Variable::GENERATED_ENERGY_MONTH, "Energy generated month", VariableDatatype::DT_FLOAT, VariableUOM::UOM_KILOWATTHOUR, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_STAT_ENERGY_GENERATED_THIS_MONTH_DF, MQTT_TOPIC_STAT_ENERGY_GENERATED_THIS_MONTH_DF);
     this->initializeVariable(Variable::GENERATED_ENERGY_YEAR, "Energy generated year", VariableDatatype::DT_FLOAT, VariableUOM::UOM_KILOWATTHOUR, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_STAT_ENERGY_GENERATED_THIS_YEAR_DF, MQTT_TOPIC_STAT_ENERGY_GENERATED_THIS_YEAR_DF);
     this->initializeVariable(Variable::GENERATED_ENERGY_TOTAL, "Energy generated", VariableDatatype::DT_FLOAT, VariableUOM::UOM_KILOWATTHOUR, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_STAT_ENERGY_GENERATED_TOTAL_DF, MQTT_TOPIC_STAT_ENERGY_GENERATED_TOTAL_DF);
-    this->initializeVariable(Variable::MAXIMUM_PV_VOLTAGE_TODAY, "Today PV max. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MIN_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MIN_PV_VOLTAGE_TODAY_DF);
-    this->initializeVariable(Variable::MINIMUM_PV_VOLTAGE_TODAY, "Today PV min. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MAX_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MAX_PV_VOLTAGE_TODAY_DF);
+    this->initializeVariable(Variable::MAXIMUM_PV_VOLTAGE_TODAY, "Today PV max. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MAX_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MAX_PV_VOLTAGE_TODAY_DF);
+    this->initializeVariable(Variable::MINIMUM_PV_VOLTAGE_TODAY, "Today PV min. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MIN_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MIN_PV_VOLTAGE_TODAY_DF);
     this->initializeVariable(Variable::MAXIMUM_BATTERY_VOLTAGE_TODAY, "Today batt. max. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MAX_BATTERY_VOLTAGE_TODAY_DF, MQTT_TOPIC_MAX_BATTERY_VOLTAGE_TODAY_DF);
     this->initializeVariable(Variable::MINIMUM_BATTERY_VOLTAGE_TODAY, "Today batt. min. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MIN_BATTERY_VOLTAGE_TODAY_DF, MQTT_TOPIC_MIN_BATTERY_VOLTAGE_TODAY_DF);
     this->initializeVariable(Variable::BATTERY_BOOST_VOLTAGE, "Boost volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READWRITE, vPIN_BATTERY_BOOST_VOLTAGE_DF, MQTT_TOPIC_BATTERY_BOOST_VOLTAGE_DF);

theaudio, 2022-11-05:
patch
diff --git a/SolarTracerBlynk/src/core/VariableDefiner.cpp b/SolarTracerBlynk/src/core/VariableDefiner.cpp
index 4e984ae..d3bfd30 100644
--- a/SolarTracerBlynk/src/core/VariableDefiner.cpp
+++ b/SolarTracerBlynk/src/core/VariableDefiner.cpp
@@ -46,8 +46,8 @@ VariableDefiner::VariableDefiner()
     this->initializeVariable(Variable::GENERATED_ENERGY_MONTH, "Energy generated month", VariableDatatype::DT_FLOAT, VariableUOM::UOM_KILOWATTHOUR, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_STAT_ENERGY_GENERATED_THIS_MONTH_DF, MQTT_TOPIC_STAT_ENERGY_GENERATED_THIS_MONTH_DF);
     this->initializeVariable(Variable::GENERATED_ENERGY_YEAR, "Energy generated year", VariableDatatype::DT_FLOAT, VariableUOM::UOM_KILOWATTHOUR, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_STAT_ENERGY_GENERATED_THIS_YEAR_DF, MQTT_TOPIC_STAT_ENERGY_GENERATED_THIS_YEAR_DF);
     this->initializeVariable(Variable::GENERATED_ENERGY_TOTAL, "Energy generated", VariableDatatype::DT_FLOAT, VariableUOM::UOM_KILOWATTHOUR, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_STAT_ENERGY_GENERATED_TOTAL_DF, MQTT_TOPIC_STAT_ENERGY_GENERATED_TOTAL_DF);
-    this->initializeVariable(Variable::MAXIMUM_PV_VOLTAGE_TODAY, "Today PV max. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MIN_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MIN_PV_VOLTAGE_TODAY_DF);
-    this->initializeVariable(Variable::MINIMUM_PV_VOLTAGE_TODAY, "Today PV min. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MAX_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MAX_PV_VOLTAGE_TODAY_DF);
+    this->initializeVariable(Variable::MAXIMUM_PV_VOLTAGE_TODAY, "Today PV max. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MAX_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MAX_PV_VOLTAGE_TODAY_DF);
+    this->initializeVariable(Variable::MINIMUM_PV_VOLTAGE_TODAY, "Today PV min. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MIN_PV_VOLTAGE_TODAY_DF, MQTT_TOPIC_MIN_PV_VOLTAGE_TODAY_DF);
     this->initializeVariable(Variable::MAXIMUM_BATTERY_VOLTAGE_TODAY, "Today batt. max. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MAX_BATTERY_VOLTAGE_TODAY_DF, MQTT_TOPIC_MAX_BATTERY_VOLTAGE_TODAY_DF);
     this->initializeVariable(Variable::MINIMUM_BATTERY_VOLTAGE_TODAY, "Today batt. min. volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READ, vPIN_MIN_BATTERY_VOLTAGE_TODAY_DF, MQTT_TOPIC_MIN_BATTERY_VOLTAGE_TODAY_DF);
     this->initializeVariable(Variable::BATTERY_BOOST_VOLTAGE, "Boost volt.", VariableDatatype::DT_FLOAT, VariableUOM::UOM_VOLT, VariableSource::SR_STATS, VariableMode::MD_READWRITE, vPIN_BATTERY_BOOST_VOLTAGE_DF, MQTT_TOPIC_BATTERY_BOOST_VOLTAGE_DF);

Bettapro, 2022-11-05:
Thanks @theaudio, I'll apply your fix.

commit af56f789 (2022-11-05): Fix #54 Today pv MIN/MAX voltage
closed 2022-11-05 by Bettapro