Bettapro/Solar-Tracer-Blynk-V3 #66: Duplicate unique IDs and sensor name using multiple device sensor on mqttHA

ntah, 2023-06-07:
**Is your feature request related to a problem? Please describe.**
Hello,
HA ignoring all new added sensor if using multiple device sensor because sensor unique ids already exists.

**Describe the solution you'd like**
We can using device name in front of variable to void multiple unique id and name
"name":  ids + "PV volt.",
"uniq_id": ids +"_pv_voltage"

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[image]

Bettapro, 2023-06-07:
This issue is related to [this pull request](https://github.com/dawidchyrzynski/arduino-home-assistant/pull/134) I created, unfortunatelly no updates on this side.

The only way to make it work(since the pull request has not been merged ) is to edit your config.h and compile the fw.
Put a unique value on  `MQTT_TOPIC_ROOT`.
Is not possibile to define this value elsewhere at the moment (eg: web config).

[image]

Bettapro, 2023-06-07:
Your solution should work but it still needs to compile the for each device. 

As I said in the pull request, the device shouldn't be identified by its "name",  we shoud provide a unique id instead, not related to its name.

I guess, the best bet are:

1. Use my branch for home-assistant-arduino (replace lib_deps with ' https://github.com/Bettapro/arduino-home-assistant.git#unique-id')
2. Add new variable `MQTT_HOME_ASSISTANT_DEVICE_ID`, that could be modified  in the web config (like `MQTT_HOME_ASSISTANT_DEVICE_NAME`), dafault value will be "" to keep it backward compatible.
3. HA-sync will prepend `MQTT_HOME_ASSISTANT_DEVICE_ID` on each sensor ID to make it unique

Or:
1. Add new variable `MQTT_HOME_ASSISTANT_DEVICE_ID`, that could be modified  in the web config (like `MQTT_HOME_ASSISTANT_DEVICE_NAME`), dafault value will be "" to keep it backward compatible.
2. HA-sync will prepend `MQTT_HOME_ASSISTANT_DEVICE_ID` on each  sensor NAME to make it unique

The last one is using the original library (that is a good thing) but it changes the "name" of the sensor depending on the device Id (which is not very good).
But, as it keeps it limited to HA-sync shouldn't be a really bad thing.

Bettapro, 2023-08-04:
Solved in v3.0.9 by using the new variable MQTT_HOME_ASSISTANT_DEVICE_ID (in config.h or web config)

closed 2023-08-04 by Bettapro