Bettapro/Solar-Tracer-Blynk-V3 #67: "Regarding ESP8266 D1 mini, I cannot find the 'SolarTracerAP' Wi-Fi access point when flashing the firmware."

ahn254, 2023-08-03:
I tried to flash "SolarTracerBlynk_3.0.8_esp8266_blynk.bin" into my "esp8266 D1 mini" board. However, after flashing, my phone cannot detect the "SolarTracerAP" Wi-Fi access point. How can I resolve this issue?

I strictly follow the instructions below for the wiring.

ESP8266 <-> MAX485 module + SOFTWARE SERIAL (since v3.0.7)
Required connections:

DI -> D2 / GPIO4
RO -> D4 / GPIO2
DE and RE are interconnected with a jumper and then connected to pin D1 / GPIO5
VCC to 3.3V on ESP8266
Alt text

MAX485 module <-> EPEVER CONTROLLER (RJ45)

ahn254, 2023-08-03:
[image]
[image]

Bettapro, 2023-08-03:
Change the offset value from 0x10000 to 0x0, then start the process again (the value next to the bin file input)
[image]

Let me know if this solves your issue, so I can update the doc.