Bettapro/Solar-Tracer-Blynk-V3 #72: Changing settings after Wifi connection unavailable

MidnightLink, 2023-08-19:
**Describe the bug**

After connecting to WIFI you are unable to login and make any additional changes

**To Reproduce**
Steps to reproduce the behavior:
1. Install SolarTracerBlynk_3.0.4_esp32dev_FULL.bin
2. Update to SolarTracerBlynk_3.0.9_esp32dev_mqttHA.bin
3. Connect to captive portal and put in correct WIFI credentials 
4. Wait for the device to connect to WIFI
5. Go to the IP address assigned to the device

**Expected behavior**
Being able to view settings, info, WIFI config page from initial portal

Bettapro, 2023-08-20:
The behaviour you described is how it's meant to work at the moment.
Config mode (web page) is not available all the time to spare resources. Normally you won't need to access to it unless  you are testing or setting it up.

You can start the config mode at any time just by double pressing the reset button or using the trigger pin or via the internal hall sensor, depending on your config.h.

Is there any reason why you need such feature?

MidnightLink, 2023-08-20:
Understood! Thank you for your time :)

closed 2023-08-20 by MidnightLink