Bettapro/Solar-Tracer-Blynk-V3 #74: Home Assistant Energy Dashboard - Long-term Statistics

MidnightLink, 2023-08-19:
**Describe the bug**
The MQTT integration doesn't report any usable device energy statistics in Home Assistant

**To Reproduce**
Steps to reproduce the behavior:

1. Set up device with SolarTracerBlynk_3.0.9_esp32dev_mqttHA

**Expected behavior**
Add device statistics to Energy Dashboard in Home Assistant

**Screenshots**
[image]

**Additional context**
Device is configured and data is coming through but I'm unable to add to the energy dashboard as shown on the HA config page. MQTT device page looks normal and I'm able to add the sensors to regular dashboards

Bettapro, 2023-08-20:
Are yuo still unable to selected energy readings? as far as I know the UOM (Kwh) should be enought.
If not please provide some more details like your config.h and your HA config page (screenshot)

MidnightLink, 2023-08-20:
config.h hasn't been changed as I'm using the default esp32dev_mqttHA file

Anything in particular you'd like a screenshot of? Like MQTT?

Bettapro, 2023-08-20:
I double check my setup, home assistant requires those value to be `state_class=total_increasing`.
The mqtt-HA library cannot specify this..... 😞 

You can do so by:
1. Editing yaml file of your HA, adding something like:
2.

MidnightLink, 2023-08-20:
I'm getting an invalid domain error when trying to add it to my yaml config.  Do you have it under sensors?

Bettapro, 2023-08-21:
Here the complete entry:
```
homeassistant:
  customize:
    sensor.solartracer1_solartracer_energy_generated:
      state_class: total_increasing
```

You can refer to HASS docs: https://www.home-assistant.io/docs/configuration/customizing-devices/#manual-customization