Bettapro/Solar-Tracer-Blynk-V3 #81: circular_queue.h: No such file or directory, after EspSoftwareSerial upgrade to 8.2.0

grz3chu, 2024-01-14:
The EspSoftwareSerial project introduced major changes in version 8.2.0, currently the project is only built in version 8.0.3.

`In file included from .pio/libdeps/wemos_d1_mini32/EspSoftwareSerial/src/SoftwareSerial.cpp:21:
.pio/libdeps/wemos_d1_mini32/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

************************************************************************
* Looking for circular_queue.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:circular_queue.h"
* Web  > https://registry.platformio.org/search?q=header:circular_queue.h
*
************************************************************************

 #include <circular_queue.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\wemos_d1_mini32\libcdf\ArduinoOTA\ArduinoOTA.cpp.o
Compiling .pio\build\wemos_d1_mini32\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\wemos_d1_mini32\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\wemos_d1_mini32\FrameworkArduino\FunctionalInterrupt.cpp.o
*** [.pio\build\wemos_d1_mini32\lib65f\EspSoftwareSerial\SoftwareSerial.cpp.o] Error 1
========================================= [FAILED] Took 18.81 seconds =========================================

Environment      Status    Duration
---------------  --------  ------------
wemos_d1_mini32  FAILED    00:00:18.806
==================================== 1 failed, 0 succeeded in 00:00:18.806 ====================================`

grz3chu, 2024-01-14:
Switch to https://github.com/plerup/espsoftwareserial.git#8.0.3 in platformio.ini solves the problem.

Bettapro, 2024-01-15:
As far as I understand, the problem is caused by a transitive dependency that is missing (platformio is not able to resolve and install at the moment).
[image]

The library is https://github.com/dok-net/ghostl .
You can use the previous version (as you mentioned) or just add it to `lib_deps` like so:
[image]

These are just  temp. solutions, it will be solved automagically as soon as the library "ghostl" will be added to platformio registry.