Bettapro/Solar-Tracer-Blynk-V3 #85: No data over mqtt after router reboot

mihai-catalin, 2024-11-06:
Hi,
I have the issue I mentioned in the title without any idea how to solve it.
After every router reboot (intentionally or not) the tracer doesn't send anything to mqtt broker until I restart the module.
I am using swver version without any change.
Must be mentioned that I can still see it connected on the router. 
Any idea what I should do?
Thank you

Bettapro, 2024-11-11:
Hello, are you able to get the LOG for the serial port? if so, please attach a complete trace ?

mihai-catalin, 2024-11-20:
> Hello, are you able to get the LOG for the serial port? if so, please attach a complete trace ? 
> 

This is in a remote location. I will try when reaching that location.
Thank you